zeitgeist87/RFReceiver #1: crc16.h no such file or directory

Castrovalva, 2017-01-15:
When uploading the example file to my MKR1000 I get this error:

```
fatal error: util/crc16.h: No such file or directory

 #include <util/crc16.h>

                        ^
```

Is this a known issue, what would be the fix?

zeitgeist87, 2017-01-16:
The header `util/crc16.h` is part of avr-libc. Since the MKR1000 uses a ARM Cortex-M0 MCU, it probably uses another libc implementation, which does not provide this header. Unfortunately I don't have a MKR1000 board.

> Is this a known issue, what would be the fix?

It is a known issue in the sense, that this library currently has no support for the new ARM based Arduino boards. The solution would be to use a different CRC16 implementation, but I suspect, that this will not be enough to make it work with the MKR1000.

However I do plan on adding support for the new boards in the future.

Castrovalva, 2017-01-16:
Thanks for your answer, it makes sense. Do you know of a RF library that works on the MKR1000/ARM Cortex-M0 MCU, both transmitting and receiving?

zeitgeist87, 2017-01-19:
I currently don't own the MKR1000 or any other ARM based board, so I haven't been looking for RF libraries. But I certainly plan to order one and update this library.

closed 2017-01-19 by zeitgeist87

zeitgeist87, 2017-02-15:
@Castrovalva I've added a fallback mode for unsupported MCUs. This should also work with the MKR1000.

Castrovalva, 2017-02-15:
@zeitgeist87 I'll try it out some day in the future. Still working on a project that uses these 433Mhz components. Right now I'm using the RC-Switch library to send over integers.