zeitgeist87/RFReceiver #3: crash, stack trace

Thijxx, 2017-04-15:
Using this library all my ESP does is crashing. It leaves a stack trace and that is it.
Tested on ESP7, ESP12 and NodeMCU v3 board. They all crash.

zeitgeist87, 2017-04-16:
It works quite well for me with NodeMCUv3 and platformio. I used the following code to test it:

```cpp
#include <Arduino.h>

#include <PinChangeInterruptHandler.h>
#include <RFReceiver.h>

RFReceiver receiver(0);

void setup() {
  Serial.begin(9600);

  receiver.begin();

  Serial.println("REBOOT");
}

void loop() {
  byte data[MAX_PACKAGE_SIZE];
  byte from = 0;
  byte packageId = 0;
  if (!receiver.ready()) {
    return;
  }

  Serial.println("Waiting");
  byte len = receiver.recvPackage(data, &from, &packageId);

  Serial.println("");
  Serial.print("Package: ");
  Serial.println(packageId);
  Serial.println("");
}
```

This is the content of my `platformio.ini`:

```ini
[platformio]
lib_dir=~/Arduino/libraries

[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino
```
The pins on the board do not directly correspond to the Arduino pins:

```cpp
static const uint8_t D0   = 16;
static const uint8_t D1   = 5;
static const uint8_t D2   = 4;
static const uint8_t D3   = 0;
static const uint8_t D4   = 2;
static const uint8_t D5   = 14;
static const uint8_t D6   = 12;
static const uint8_t D7   = 13;
static const uint8_t D8   = 15;
static const uint8_t D9   = 3;
static const uint8_t D10  = 1;
```

I used the pin marked `D3` on the board, which corresponds to Arduino pin 0. This is the serial output:

```
Package: 79

Waiting

Package: 80

Waiting

Package: 81
```

Thijxx, 2017-04-16:
Thank for the great reply! I will retry this using your verified working code as soon as possible.