zeitgeist87/RFReceiver #5: Can't receive packets from unmarked chip

Nihlus, 2017-11-20:
I have a small 433mhz keyfob with an unmarked chip, and I'm trying to receive data packets from it so that I can retransmit them later. Unfortunately, I can't seem to get any library to recognize it. I know that it's 433 mhz since the clock crystal on the board is marked.

I've tried both the blocking and non-blocking versions of the test code. I know that my receiver is getting something (I've temporarily hooked up an LED to the data pin for debugging purposes, and I can clearly see a flash when I use the keyfob). However, no packets are recognized.

Any pointers on what to look for or what to try? This is my first foray into RF transmission and reception with the Arduino, so I am a little lost.

zeitgeist87, 2017-11-21:
Hi @Nihlus,

The RFReceiver can only receive packets from the RFTransmitter library, because both speak the same protocol. You need a library that understands the protocol your keyfob speaks. It is not just about the packet format, but also about how the individual bits are encoded. Unfortunately, I have no experience with these keyfobs and can't tell you which library will work.

Nihlus, 2017-11-21:
Ah, I see - thank you for the explanation. I did eventually manage to figure it out with a bit of probing around, and it turns out my fob was a bit of an oddball anyway. I used rc-switch in the end with a custom protocol.

All is well :)

closed 2017-11-21 by Nihlus