zeitgeist87/RFReceiver #7: Attiny85 problem?

lazlyhu, 2018-04-17:
Hi!

Thank you for your work, I really like it, just working. :)
I have some change, I'm planning to share here but first I wanted to finish my project.

I'm ready with my project (in software side) and made a PCB for finalise it in an Attiny85. In probe panel with Arduino (be exact, on the receiver side an Arduino Nano clone) everything worked great, I attached the the RX radio to pin2, working great.
Because Pin2 in Arduino is `INT0` I decided to use the `INT0` in Attiny too, so used PB1. After the PCB went ready I recognised my mistake, because PB1 is `INT0` only in attiny13, in Attiny85 I should use PB2. PB2 was free, so now I connected it to PB1.

Anyway, my issue looks like because some reason It's not working on any port (PB1 or PB2). Do you have any idea why? I complied with DIY AtTiny, and first write out the bootloader to setup the fuses. Maybe some fuse settings failed?

Thank you, wish the best!

lazlyhu, 2018-04-17:
Looks like should an issue around `PinChangeInterruptHandler` lib, or I missed something in that part of the code. I wrote two example sketch. The first working, the second don't. What do you think?

Uploaded here: https://gist.github.com/lazlyhu/e6a309348ca658d63e38f8cd66d2cee8

lazlyhu, 2018-04-24:
I don't have any idea what's wrong with `PinChangeInterruptHandler` but there should be some issue. I really want start using the result of my product, so I rewrite `RFReceiver` out of `PinChangeInterruptHandler` based on this idea:

Base steps:
- Practically you don't need `receiver.begin();` anymore, don't need to call
- But in `setup()` you should call something like: `attachInterrupt(digitalPinToInterrupt(SIGNAL_PIN), interupt, CHANGE);`
- And define somewhere a simple target of the interrupt: `void interupt() {
  receiver.handlePCInterrupt();
}`
- Refactor the `RFReceiver` to recive receiver.handlePCInterrupt(); as a public call without any argument, drop the included `PinChangeInterruptHandler`, etc

I know, It's not solve the problem. Maybe if you have time feel free to reopen the issue and I would be happy to help you find some real solution. But I left here, maybe years later somebody looking some fast solution too. 😀

closed 2018-04-24 by lazlyhu

zeitgeist87, 2018-04-25:
Hi @lazlyhu,

I am sorry that I didn't respond to your issue. I am currently extremely busy at work. I have an Attiny85 in my junkbox. I will try to reproduce the issue this weekend.

lazlyhu, 2018-04-25:
@zeitgeist87

It's fine, don't worry :) Let me know if I can help you to fix this.

nizarmhd, 2021-09-06:
> @zeitgeist87
> 
> It's fine, don't worry :) Let me know if I can help you to fix this.

hello, did you fix this issue...?