zeitgeist87/RFReceiver #9: stopping the searching of new messages

MassiStr, 2020-04-11:
Hi, nice work! I have a problem: how can I stop searching for a new message from the transmitter, and go on with the sketch, for example because the transmitter is now offline and i have other sensors to check after? for example, in VirtualWire I have the command "vW wait ()"..Thanks!!

zeitgeist87, 2020-04-12:
Hi @MassiStr,

The receiver has a method called `ready()`. If `ready()` returns `true`, then a subsequent call to `recvPackage()` won't block. So the code would look like this:

```cpp
if (receiver.ready()) {
    receiver.recvPackage(...);
    // Do something with the data
} else {
    // Check your other sensors
}
```

Regards,
Andreas

MassiStr, 2020-04-13:
Hi zeitgeist87, 
THANKS, It works perfectly!
I found your library very useful, It works even with the new Arduino Nano Every (Atmega4809), the VirtualWire/RadioHead did not works.

Regards, 
Massimo