zeitgeist87/RFReceiver #10: Odd data

Tempy111, 2020-06-19:
I'm using a Sparkfun Pro Micro which uses a ATmega32U4 chip,
Following the code in the example for the RF receiver and transmitter, I'm successfully transmitting but the receiver is just picking all the signal's as "T".

for transmitting the code I'm using is:
```
 char msg[] = "HR";
 transmitter.send((byte *)msg, strlen(msg) +1);
```

(the char msg line is slightly different from the example cause I was getting some warnings, but when I tried the original, it didn't make any visual difference apart from the warnings). I do a serial print line to check the msg value..

here is the full code (so far really.) for the receiver:
```
#include <Servo.h>  // servo library
#include <PinChangeInterruptHandler.h>
#include <RFReceiver.h>
#include <SPI.h> //Needed to compile?

Servo servoArmL;         // servo control object for the Left Arm
Servo servoArmR;         //Servo control object for the Right Arm
Servo servoHead;          // Servo control for the head

//RFTransmitter transmitter(OUTPUT_PIN, NODE_ID);
RFReceiver receiver(9); //Receive on Pin 2
//Receiver cannot be pin 2.. needs to be Pin 8,9,10,11,14,15 or 16 on a pro micro 

int buttonState = 0;      //variable for reading the pushbutton status
int RHeadState = 0;
int LHeadState = 0;

int HeadPos = 90;
int armpos = 0;           //variable for roughtly where the arms are. 0 is down, 1 is up
boolean actionDone = true;//This will tell when an action is done, and there for the next
                          //action can be set
void setup()
{
  servoArmR.attach(15);//, 900, 2100);  //Connect the servo on the Left arm to pin 6
  servoArmL.attach(7); //Connect the servo on the Right arm to pin 5
  servoHead.attach(20); //6 //Connect head Servo to pin 6;
  //Initialize the pushbutton pin as an input:
  servoArmR.write(0);   //Might be an idea to automatically reset the arm position to down
  servoArmL.write(90);
  servoHead.write(90);
  Serial.begin(9600);
  receiver.begin();
}
void loop()
{
  char msg(MAX_PACKAGE_SIZE);
  byte senderId = 0;
  byte packageId = 0;
  byte len = receiver.recvPackage((byte *)msg, &senderId, &packageId);

  Serial.println(msg);
  int position;
  int HeadPos = 90;
  if (msg =="ArmUp"){
    servoArmR.write(90);
    servoArmL.write(0);
    delay(100);
  } else if (msg == "ArmDown"){
    servoArmR.write(0);
    servoArmL.write(90);
    delay(100);
  } else if (msg == "HReset"){
    servoHead.write(90);
    delay(20);
  } else if (msg == "HR"){
    if (HeadPos > 0){
      HeadPos -= 2;
    }
    servoHead.write(HeadPos);
    delay(20);
  } else if (msg == "HL"){
    if (HeadPos < 180){
      HeadPos += 2;
    }
    servoHead.write(HeadPos);
    delay(20);
  }
}
```

zeitgeist87, 2020-06-20:
Hi @Tempy111,

At first glance, you used the wrong kind of brackets for the char array. Try this:

```cpp
char msg[MAX_PACKAGE_SIZE];
```

Also you cannot do a string comparison using the `==` operator in C++. At least not using char pointers. Try this:

```cpp
if (strncmp(msg, "ArmUp", sizeof("ArmUp")) == 0){
...
}
```

Here is a reference for the `strncmp()` function: http://www.cplusplus.com/reference/cstring/strncmp/

Alternatively you can try to convert the msg array into an object of the `String` class: https://www.arduino.cc/reference/en/language/variables/data-types/stringobject/

```cpp
// You always need to make sure that msg is 0-terminated
msg[MAX_PACKAGE_SIZE - 1] = 0;
String msg_copy(msg);

if (msg_copy == "ArmUp") {
...
}
```

Tempy111, 2020-06-20:
Ah.. thanks.. my C++ is very rusty..

I'll have to test that more a bit later cause I just tried and I wasn't getting any signal transmitted.. so i'll have to check a bit later cause that's an unrelated issue that needs dealing with ^_^

Tempy111, 2020-06-20:
all works now... but.. the servo keeps twitching.. not sure why.. not getting any message on the serial output, which I set up so that i'll get a message when a servo action is done... and there are no crossed wires... weird.. can't quite see how the RF transmitter is interfering like that..